chipsalliance/treadle #62: VCD dumps not generated in certain conditions

ducky64, 2018-11-16:
It seems single-cycle runs won't generate a VCD dump.

Also, `report()` needs to be called, which isn't obvious and also includes console printing. Perhaps separate that functionality into finish / write_vcd / report_statistics (which would return a case class, leaving it to the testdriver to report it)?

chick, 2018-11-29:
@ducky64 Is this still an issue? I agree with a cleanup but it seems like this is really a general Chisel simulation API issue, and the specifics of treadle can be changed to be in line with whatever we need.

chick, 2018-11-30:
@ducky64 can you point me at an example. I tried to reproduce locally in treadle but could not.

closed 2019-02-12 by chick