chipsalliance/treadle #78: xorR is broken in chisel-tester?

zinechant, 2019-01-17:
Type of issue: bug report

From: https://github.com/freechipsproject/firrtl/issues/993

I am new to chisel world. While I am exploring, I noticed that the xorR is broken in firrtl backend for unit tests.

There were discussions about broken of reduction operators. https://github.com/freechipsproject/firrtl/issues/356
But it seems xorR is still tied to the lowest bit.

A simple [repo](https://github.com/zinechant/xor_reduce) testing xorR.

Regards,
Chen

chick, 2019-01-17:
Thanks for the report, this seems to be a treadle only bug, I have created PR #79 to fix it.

zinechant, 2019-01-17:
Thanks.
It seems the default backend for iotester.Driver is already treadle.
I confirm that firrtl backend actually gives the correct behavior of xorR.

closed 2019-02-12 by chick