chipsalliance/treadle #85: Treadle gives poor performance with large memory

sashimi-yzh, 2019-02-12:
I use [this version of chisel-template](https://github.com/freechipsproject/chisel-template/tree/014630fbcf5a8caf082a9c0fd699d7486f4d2429) to develop a single-cycle CPU. I use the following statement to create a memory which the CPU is going to access.
```
val mem = Mem(memByte, UInt(8.W))
```
I run chisel-tester with the `treadle` backend. The tester is something like
```
do {
  step(1)
  state = peek(top.io.state)
} while (state != END)
```

**But I observe that when `memByte` grows, `treadle` gives poor performance.**

I start by running my CPU with a small program. This program require 64KB memory to run. And it ends after about `120000` cycles. Then I double the value of `memByte`, and run the program again. Note that doubling the memory size does not affect the program behavior. I collect the real world time cost during simulation, and the simulation frequency. This data is reported by `treadle` after the simulation ends.

```
===== 120000 cycles  (-J-Xmx512M -J-Xms512M) =====

64KB -> 9.294537 seconds 12848.41 Hz
128KB -> 12.573144 seconds 9506.77 Hz
256KB -> 23.733160 seconds 5043.37 Hz
512KB -> 61.044713 seconds 1962.58 Hz
1MB -> 144.135788 seconds 833.48 Hz
```
This shows that when the size of memory doubles, the real world time also doubles, meanwhile the frequency drops to half.

When the memory size grows to 1MB, it takes a long time to finish the program. Therefore I modify the tester to let the simulation end after `12000` cycles:
```
do {
  step(1)
  state = peek(top.io.state)
  nrInstr +=1
} while (state != END && nrInstr < 12000)
```
Then I continue to collect data.
```
===== 12000 instructions (-J-Xmx512M -J-Xms512M) =====

1MB -> 15.067397 seconds 796.75 Hz
2MB -> 29.217526 seconds 410.88 Hz
4MB -> 57.313398 seconds 209.46 Hz
8MB -> 118.053021 seconds 101.69 Hz
```

Again, when the memory size grows to 8MB, it takes about 2 minutes to finish the simulation. I further reduce the number of simulation cycles.
```
===== 1200 instructions (-J-Xmx512M -J-Xms512M) =====

8MB -> 12.139967 seconds 99.26 Hz
16MB -> "java.lang.OutOfMemoryError: Java heap space"
```
But when I set the memory size to 16MB, I get the following exception.
```
[info] [0.002] Elaborating design...
[info] [1.397] Done elaborating.
Total FIRRTL Compile Time: 1428.6 ms
Total FIRRTL Compile Time: 574.2 ms
file loaded in 0.935446212 seconds, 734 symbols, 686 statements
[error] (run-main-0) java.lang.OutOfMemoryError: Java heap space
[error] java.lang.OutOfMemoryError: Java heap space
[error]         at scala.collection.mutable.ArrayBuilder$ofInt.mkArray(ArrayBuilder.scala:335)
[error]         at scala.collection.mutable.ArrayBuilder$ofInt.resize(ArrayBuilder.scala:341)
[error]         at scala.collection.mutable.ArrayBuilder$ofInt.sizeHint(ArrayBuilder.scala:346)
[error]         at scala.Array$.fill(Array.scala:264)
[error]         at treadle.executable.RollBackBuffer.<init>(RollBackBuffer.scala:10)
[error]         at treadle.executable.RollBackBufferRing$$anonfun$4.apply(RollBackBuffer.scala:24)
[error]         at treadle.executable.RollBackBufferRing$$anonfun$4.apply(RollBackBuffer.scala:24)
[error]         at scala.Array$.fill(Array.scala:267)
[error]         at treadle.executable.RollBackBufferRing.<init>(RollBackBuffer.scala:24)
[error]         at treadle.executable.RollBackBufferManager.saveData(RollBackBuffer.scala:69)
[error]         at treadle.executable.DataStore.saveData(DataStore.scala:117)
[error]         at treadle.executable.SimpleSingleClockStepper$$anonfun$run$1.apply$mcVI$sp(ClockStepper.scala:176)
[error]         at scala.collection.immutable.Range.foreach$mVc$sp(Range.scala:160)
[error]         at treadle.executable.SimpleSingleClockStepper.run(ClockStepper.scala:150)
[error]         at treadle.TreadleTester.step(TreadleTester.scala:269)
[error]         at chisel3.iotesters.TreadleBackend.reset(TreadleBackend.scala:113)
[error]         at chisel3.iotesters.TreadleBackend.<init>(TreadleBackend.scala:19)
[error]         at chisel3.iotesters.setupTreadleBackend$.apply(TreadleBackend.scala:144)
[error]         at chisel3.iotesters.Driver$$anonfun$execute$1$$anonfun$apply$mcZ$sp$1.apply$mcZ$sp(Driver.scala:54)
[error]         at chisel3.iotesters.Driver$$anonfun$execute$1$$anonfun$apply$mcZ$sp$1.apply(Driver.scala:39)
[error]         at chisel3.iotesters.Driver$$anonfun$execute$1$$anonfun$apply$mcZ$sp$1.apply(Driver.scala:39)
[error]         at logger.Logger$$anonfun$makeScope$1.apply(Logger.scala:138)
[error]         at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error]         at logger.Logger$.makeScope(Logger.scala:136)
[error]         at chisel3.iotesters.Driver$$anonfun$execute$1.apply$mcZ$sp(Driver.scala:39)
[error]         at chisel3.iotesters.Driver$$anonfun$execute$1.apply(Driver.scala:39)
[error]         at chisel3.iotesters.Driver$$anonfun$execute$1.apply(Driver.scala:39)
[error]         at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error]         at chisel3.iotesters.Driver$.execute(Driver.scala:38)
[error]         at chisel3.iotesters.Driver$.execute(Driver.scala:100)
[error]         at top.TestMain$.delayedEndpoint$top$TestMain$1(NOOPTester.scala:128)
[error]         at top.TestMain$delayedInit$body.apply(NOOPTester.scala:120)
[error] java.lang.RuntimeException: Nonzero exit code: 1
[error]         at sbt.Run$.executeTrapExit(Run.scala:124)
[error]         at sbt.Run.run(Run.scala:77)
[error]         at sbt.Defaults$.$anonfun$bgRunMainTask$6(Defaults.scala:1147)
[error]         at sbt.Defaults$.$anonfun$bgRunMainTask$6$adapted(Defaults.scala:1142)
[error]         at sbt.internal.BackgroundThreadPool.$anonfun$run$1(DefaultBackgroundJobService.scala:366)
[error]         at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error]         at scala.util.Try$.apply(Try.scala:209)
[error]         at sbt.internal.BackgroundThreadPool$BackgroundRunnable.run(DefaultBackgroundJobService.scala:289)
[error]         at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error]         at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error]         at java.lang.Thread.run(Thread.java:748)
[error] (Test / runMain) Nonzero exit code: 1
```

This is strange. I have set the JVM memory to 512MB, but a 16MB `Mem` circuit cause `treadle` to exhaust all the JVM memory. I do not understand why. I set the JVM memory to 2GB and continue.
```
===== 1200 instructions (-J-Xmx2G -J-Xms2G) =====

16MB -> 24.016523 seconds 50.17 Hz
32MB -> 47.499757 seconds 25.37 Hz
64MB -> 94.620383 seconds 12.74 Hz
128MB -> "java.lang.OutOfMemoryError: Java heap space"
```
All the data above shows the same trend.

But when I run the chisel-tester with the `verilator` backend, it costs about 2 seconds to finish the whole program (about 120000 cycles), even though I set `memByte` to 128MB!
```
Exit Code: 0
[info] [2.151] RAN 119195 CYCLES PASSED
```
With simple calculation, the simulation frequency of `verilator` is about 55314 Hz, much more than `treadle`.

chick, 2019-02-12:
Thanks for the report. Treadle is mostly focused on smaller unit tests where it offers fast spin-up and modest performance, with some additional debugging capabilities. But we do want treadle to be as useful as possible so I will look into to this further, I am not sure why the larger memory is slowing it down. I will also check out what the limitations and specs for memory sizes should be.

chick, 2019-02-13:
@sashimi-yzh The very large memories are still a problem but I have made a change in PR #87 that should fix the bad performance over working memory sizes. You can change an option in your current tests that should work. There is an internal experimental configuration option called rollbackBuffers that should be set to 0. You can do this by creating a options manager and overriding the treadleOptions with 

`rollbackBuffers = 0`

 or using the command line style and using 

`Array("--tr-rollback-buffers", "0")`

sashimi-yzh, 2019-02-14:
After adding this option, I can run the CPU with 128MB `Mem` successfully.

But the performance seems worse than `verilator`. To compare, I run another test program on my CPU. This test program require about 670000 cycles to finish. Below is the performance data.
```
=== treadle ===
670488 cycles in 22.886233 seconds 29296.56 Hz

=== verilator ===
[info] [6.506] RAN 670428 CYCLES FAILED FIRST AT CYCLE 670428
```
The result shows that `treadle` is about 3X slower than `verilator` on this test program.

jackkoenig, 2019-02-14:
Treadle is not intended to be as fast as Verilator--Verilator is a highly optimized Verilog to C++ compiler while Treadle is an interpreter. The goal of Treadle is to be fast enough for small tests and providing additional functionality (like the repl) while having a relatively quick startup time (whereas Verilator requires running Verilator and then a C++ compiler).

sashimi-yzh, 2019-02-14:
Actually, I tried to used the scala `Array` in the Tester to simulate the memory outside the chisel code several days ago. At that time, `verilator` is 1.5X slower than `treadle`. But now when I move the memory inside the chisel code, `treadle` is 3X slower than `verilator` instead. This is the reason why I wonder there may be some performance bugs while `treadle` interprets the `Mem` circuit.

chick, 2020-07-15:
This should be re-opened later if necessary

closed 2020-07-15 by chick