chipsalliance/treadle #101: Add support for FST output

ekiwi, 2019-05-20:
The `fst` file format is supported by GTKwave and has the following nice properties:
* it's compressed, so the files are much smaller than `VCD` files
* it supports SV records and enums, so there is potential to use some high level info from Chisel in order to make the output nicer

The biggest downside is the terrible documentation. The best material I have found so far is the paper at the end of the [GTKwave manual](http://gtkwave.sourceforge.net/gtkwave.pdf) titled _Implementation of an Efficient Method for Digital Waveform Compression_. Also the source code from icarus verilog is fairly readable: [sys_fst.c](https://github.com/steveicarus/iverilog/blob/11c826216a63cfeefe05f79f7023f669ef6e962a/vpi/sys_fst.c), [fstapi.h](https://github.com/steveicarus/iverilog/blob/11c826216a63cfeefe05f79f7023f669ef6e962a/vpi/fstapi.h) and [fstapi.c](https://github.com/steveicarus/iverilog/blob/11c826216a63cfeefe05f79f7023f669ef6e962a/vpi/fstapi.c).

chick, 2019-05-21:
Do you have fst files that you would like to be able to process? I looked through the appendix and it seem comprehensible, but a bit elaborate with the database and compression etc. Treadle's basic speed is not that fast so typically any vcd data one is working with is usually not so big as to be prohibitive to process. If I was making more progress on the chisel GUI it might be important because passing wave data from treadle to a rendering agent could be a problem, but there it would probably be better to have a different organization. I'd be happy to talk to you more about this if you'd like

ekiwi, 2019-05-21:
> I'd be happy to talk to you more about this if you'd like

Sounds good. I mostly want to have a more expressive waveform view.

Let's consider this a tracking issue for now. If I have time after the retreat, I could try to implement a fst writer in Scala.

chick, 2019-05-21:
Sounds good to me.

Timmmm, 2024-02-15:
I wrote a specification for FST [here](https://blog.timhutt.co.uk/fst_spec/). I also wrote some incomplete code to read it [here](https://github.com/Timmmm/wavery/tree/master/fst/src) and the [Surfer project](https://surfer-project.org/) also supports reading FST (code [here](https://github.com/ekiwi/wellen)). Their code looks way better than mine tbh.

Should give a good idea how to write FST files.

ekiwi, 2024-02-15:
> I wrote a specification for FST [here](https://blog.timhutt.co.uk/fst_spec/).

Thanks for writing that spec Tim! I actually had a look at it when I was implementing the FST reader for Surfer. (The reader is published as a stand alone library btw: https://crates.io/crates/fst-native).

Unfortunately I never quite finished making a FST writer. The hard part is imho how to efficiently store the change data as it comes from the simulator before reformatting it to write it to disk. As I like to say: VCD is easy to write, hard to read (efficiently). FST is easy to read, but hard to write (efficiently).

Alas, unfortunately `treadle` has been mostly abandoned. I keep a version of it alive as part of my [chiseltest](https://github.com/ucb-bar/chiseltest) library, but unfortunately I won't have time to add new features.