chipsalliance/treadle #124: concat bits wrong!

shady831213, 2019-08-16:
Problem:
Here's the logic:
```
reg fb_uop_ram_0_inst : UInt<32>, clock with
reg fb_uop_ram_0_pc : UInt<32>, clock with 

node _T_29 = cat(cat(fb_uop_ram_0_pc, fb_uop_ram_0_inst), fb_uop_ram_0_error)

pipeline.FetchBuffer.fb_uop_ram_0_inst   Lon UInt     32      1    141L   1119 
pipeline.FetchBuffer.fb_uop_ram_0_pc     Lon UInt     32      1    238L   1121 

pipeline.FetchBuffer._T_29               Big UInt     65      1      2B   1122 
```

when fb_uop_ram_0_inst = 0xF1402573, fb_uop_ram_0_pc = 0x8000004C, _T_29 get wrong result(0x1FFFFFFFFE2804AE6)
Here's the waveform:
[image]

Analysis:
Here's [CatBigs](https://github.com/freechipsproject/treadle/blob/cf1cfad1f2a94d28c6eab5d536a18724668043ab/src/main/scala/treadle/executable/BigPrimOps.scala#L190) in :
```scala
case class CatBigs(f1: FuncBig, f1Width: Int, f2: FuncBig, f2Width: Int) extends BigExpressionResult {
  private val mask1 = BitMasks.getBitMasksBigs(f1Width).allBitsMask
  def apply(): Big = {
    ((f1() & mask1) << f2Width) | f2()
  }
}
```
f2() is not masked. Because fb_uop_ram_0_inst is type of Long as above, before calling CatBigs, it  have to be covert to BigInt As [here](https://github.com/freechipsproject/treadle/blob/cf1cfad1f2a94d28c6eab5d536a18724668043ab/src/main/scala/treadle/executable/ExpressionCompiler.scala#L357):
```scala
        case (BigSize, e1: LongExpressionResult, e2: LongExpressionResult) =>
          handleBigResult(LongToBig(e1.apply), LongToBig(e2.apply))
```
and [LongToBig](https://github.com/freechipsproject/treadle/blob/cf1cfad1f2a94d28c6eab5d536a18724668043ab/src/main/scala/treadle/executable/IntOps.scala#L23) just call BigInt
```scala
case class LongToBig(f: FuncLong) extends BigExpressionResult {
  def apply(): Big = BigInt(f())
}
```
Because there only signed type in scala, it will be signed extended, that's the problem!
```scala
          println(f"long to big ${BigInt(0xF1400000)}%x")
          println(f"long to big ${BigInt(0x71400000)}%x")
```
and result
```
[[35minfo[0m] [0.022] long to big -ec00000
[[35minfo[0m] [0.022] long to big 71400000
```

shady831213, 2019-08-16:
I got the problem, it is in one of my scalar black box ....
Close it.

shady831213, 2019-08-16:
Although it's my fault, but if the lower bits masked in Cat, may be more secure I think ...

chick, 2019-08-17:
@shady831213 thanks for finding the bug and doing a great job identifying the source of the problem. I believe PR #125 will fix it. Nice work

shady831213, 2019-08-17:
Tahnks @chick .
Actually I don't think it's a bug, but fix it will be more secure ^_^

closed 2019-08-22 by shady831213