chipsalliance/treadle #176: RegInit not working correctly in treadle even when callResetAtStartUp = false

claford-v-lawrence, 2020-02-27:
Right now, treadle seems to automatically reset things. All the values in RegInit don't work.

chick, 2020-02-27:
Treadle initializes all components to zero. But it should be setting registers appropriately on reset. Can you provide a small firrtl example where this is failing?

claford-v-lawrence, 2020-02-28:
Basically, we are using a TreadleTester to run our simulations with the default configs managed by TreadleOptionsManager. Yet we find that our regInits doesn't really work as expected. I tried to explicitly do a reset with treadle REPL, setting initial reset to true, but I keep getting wrong result. I'm sorry I can't immediately give you a snippet because I don't really know how to single the part out. Could you perhaps give me some advice on that? Any thoughts would be welcome
Also, on a separate note, I was wondering where I could find some documentation of treadle. Mainly we are just trying to read examples from other projects, so maybe we got that part wrong.

chick, 2020-02-28:
Here's an example circuit that runs with treadle and it illustrates a register that is set to 10.U both on initial reset and manual reset. 
```scala
import chisel3._
import chisel3.tester._
import org.scalatest.{FreeSpec, Matchers}

class ResetRegTest extends FreeSpec with Matchers with ChiselScalatestTester {

  class HasReg extends MultiIOModule {
    val counter = IO(Output(UInt(16.W)))

    val cycles = RegInit(10.U(24.W))
    cycles := cycles +% 1.U

    printf("cycles %d\n", cycles)

    counter := cycles
  }

  "counter starts at 10 after initial and later reset" in  {
    test(new HasReg) { dut =>
    
      dut.counter.expect(10.U)

      dut.clock.step()
      dut.counter.expect(11.U)
      dut.clock.step()
      dut.counter.expect(12.U)
      dut.clock.step()
      dut.counter.expect(13.U)

      // manual reset
      dut.reset.poke(true.B)
      dut.clock.step()
      dut.reset.poke(false.B)

      dut.counter.expect(10.U)
    }
  }
}

```

chick, 2020-02-28:
There's not a lot of documentation on treadle, but there's a bit of the repo's wiki that describes some of the features. Is there anything in particular about treadle that you would like to know.

claford-v-lawrence, 2020-03-02:
Our snippet looks like this roughly. Sorry I couldn't just show you the code, it's still a private repo and we're still in early development stage. Here is what it looks like
```scala
import chisel3._
import chisel3.tester._
import org.scalatest.{FreeSpec, Matchers}

  class HasReg extends MultiIOModule {
    val counter = IO(Output(UInt(16.W)))

    val cycles = RegInit(10.U(24.W))
    cycles := cycles +% 1.U

    printf("cycles %d\n", cycles)

    counter := cycles
  }

 object tester {
  def build(optionsManager: SimulatorOptionsManager): String = {
    optionsManager.firrtlOptions = optionsManager.firrtlOptions.copy(compilerName = "low")

    chisel3.Driver.execute(optionsManager, () => new HasReg) match {
      case ChiselExecutionSuccess(Some(_), _, Some(firrtlExecutionResult)) =>
        firrtlExecutionResult match {
          case firrtl.FirrtlExecutionSuccess(_, compiledFirrtl) =>
            compiledFirrtl
          case firrtl.FirrtlExecutionFailure(message) =>
            throw new Exception(s"FirrtlBackend: Compile failed. Message: $message")
        }
      case _ =>
        throw new Exception("Problem with compilation")
    }

def main(args: Array[String]): Unit{
  val optionsManager = new TreadleOptionsManager()
  optionsManager.treadleOptions.copy(callResetAtStartUp = true)
  val compiledFirrtl = build(optionsManager)
  val sourceAnnotation = FirrtlSourceAnnotation(compiledFirrtl)
  val simulator = TreadleTester(compiledFirrtl, optionsManager)
  while (True){
  simulator.step(1)
  if (simulator.peek(HasReg.counter) == 14) {break}
}

}
  }
```
I'm not sure the code above will compile, but that's roughly what we're doing. In this precise scenario, the register always hold value zero. I'm not sure either whether this is a bug because we didn't use it correctly or a subtle bug hidden here. Thanks a lot for helping!

chick, 2020-03-02:
Hmm, I'm not sure why it is behaving that way. I have fiddled with your example a bit and have this version that looks to me like the register is being initialized properly. I've fixed a couple apparent problem with the way the options were initialized and shown an example of how to see treadle's symbol names. I have to head off to a meeting now but if this is helpful I could refine and modernize this example a bit more. Please take a look and let me know.
```scala

import chisel3._
import chisel3.iotesters.TesterOptionsManager
import chisel3.tester._
import firrtl.stage.FirrtlSourceAnnotation
import org.scalatest.{FreeSpec, Matchers}
import treadle.{TreadleOption, TreadleOptionsManager, TreadleTester}

class HasReg extends MultiIOModule {
  val counter = IO(Output(UInt(16.W)))

  val cycles = RegInit(10.U(24.W))
  cycles := cycles +% 1.U

  printf("cycles %d\n", cycles)

  counter := cycles
}

object RegInitProblem {
  def build(optionsManager: TesterOptionsManager): String = {
    optionsManager.firrtlOptions =
      optionsManager.firrtlOptions.copy(compilerName = "low")

    chisel3.Driver.execute(optionsManager, () => new HasReg) match {
      case ChiselExecutionSuccess(Some(_), _, Some(firrtlExecutionResult)) =>
        firrtlExecutionResult match {
          case firrtl.FirrtlExecutionSuccess(_, compiledFirrtl) =>
            compiledFirrtl
          case firrtl.FirrtlExecutionFailure(message) =>
            throw new Exception(
              s"FirrtlBackend: Compile failed. Message: $message"
            )
        }
      case _ =>
        throw new Exception("Problem with compilation")
    }
  }

  def main(args: Array[String]): Unit = {
    // fixed the use of copy here.
    val optionsManager = new TesterOptionsManager {
      treadleOptions = treadleOptions.copy(callResetAtStartUp = true)
    }
    val compiledFirrtl = build(optionsManager)
    val simulator = TreadleTester(compiledFirrtl, optionsManager)
    var done = false
    
    // example of how to look at treadle's view of the symbol table
    simulator.engine.symbolTable.nameToSymbol.keys.toSeq.sorted.foreach { key =>
      println(s"symbol: $key")
    }
    while (!done) {
      simulator.step(1)
      if (simulator.peek("counter") == 14) {
        done = true
      }
    }
  }
}
```

claford-v-lawrence, 2020-03-03:
Thanks! That indeed solves our problem. I can't express how grateful I am for your help! I have a small problem regarding all the option settings of treadle: how to set them ( and subsequently copy them ) in the right fashion? BTW, the part about listing the symbol table is super helpful... I used to elaborate it into Verilog and look at the Verilog file and the firrtl file to try and guess the symbol names. Thanks for making my day!

chick, 2020-03-04:
My pleasure to help. 

Another way to have access to the symbols is in the TreadleRepl. It has
commands to display the symbols and even dependencies.

Here's a little hint on how to use string flags in a treadle test without having to know all the underlying annotations
```scala
class MyTesterShell extends Shell("my-tester") with FirrtlCli

class MyTests extends FreeSpec {
  "this is some test" in {
    val startupAnnotations =
      (new MyTesterShell).parse(Array("--target-dir", "my-test-dir"))
    println(s"startup annos\n${startupAnnotations.mkString("\n")}")
    // You can then use these annos to pass in settings
    // treadle is written as a registered library so it's flags are automatically added in
    // firrtl flags were added manually
  }

  "show all the options" in {
    // use this to see what the available options are
    (new MyTesterShell).parse(Array("--help"))
  }
}
```

chick, 2020-03-04:
Can we close this issue now?

claford-v-lawrence, 2020-03-05:
Yes, thank you!

claford-v-lawrence, 2020-04-02:
Hi, I'm re-opening this issue to ask if there is a way to generate a VCD-dump for the preceding code example. I make added the generate vcd dump option to the code, but it seems there is no vcd file generated?
```scala
import chisel3._
import chisel3.iotesters.TesterOptionsManager
import chisel3.tester._
import firrtl.stage.FirrtlSourceAnnotation
import org.scalatest.{FreeSpec, Matchers}
import treadle.{TreadleOption, TreadleOptionsManager, TreadleTester}

class HasReg extends MultiIOModule {
  val counter = IO(Output(UInt(16.W)))

  val cycles = RegInit(10.U(24.W))
  cycles := cycles +% 1.U

  printf("cycles %d\n", cycles)

  counter := cycles
}

object RegInitProblem {
  def build(optionsManager: TesterOptionsManager): String = {
    optionsManager.firrtlOptions =
      optionsManager.firrtlOptions.copy(compilerName = "low")

    chisel3.Driver.execute(optionsManager, () => new HasReg) match {
      case ChiselExecutionSuccess(Some(_), _, Some(firrtlExecutionResult)) =>
        firrtlExecutionResult match {
          case firrtl.FirrtlExecutionSuccess(_, compiledFirrtl) =>
            compiledFirrtl
          case firrtl.FirrtlExecutionFailure(message) =>
            throw new Exception(
              s"FirrtlBackend: Compile failed. Message: $message"
            )
        }
      case _ =>
        throw new Exception("Problem with compilation")
    }
  }

  def main(args: Array[String]): Unit = {
    // fixed the use of copy here.
    val optionsManager = new TesterOptionsManager {
      treadleOptions = treadleOptions.copy(callResetAtStartUp = true)
//Line changed here
      treadleOptions = treadleOptions.copy(writeVCD = true)
    }
    val compiledFirrtl = build(optionsManager)
    val simulator = TreadleTester(compiledFirrtl, optionsManager)
    var done = false
    
    // example of how to look at treadle's view of the symbol table
    simulator.engine.symbolTable.nameToSymbol.keys.toSeq.sorted.foreach { key =>
      println(s"symbol: $key")
    }
    while (!done) {
      simulator.step(1)
      if (simulator.peek("counter") == 14) {
        done = true
      }
    }
  }
}
```

claford-v-lawrence, 2020-04-02:
My bad. After some digging through the source code, I need to call `simulator.report` or `simulator.finish`
at the end. For future reference, the code should look like
```scala
import chisel3._
import chisel3.iotesters.TesterOptionsManager
import chisel3.tester._
import firrtl.stage.FirrtlSourceAnnotation
import org.scalatest.{FreeSpec, Matchers}
import treadle.{TreadleOption, TreadleOptionsManager, TreadleTester}

class HasReg extends MultiIOModule {
  val counter = IO(Output(UInt(16.W)))

  val cycles = RegInit(10.U(24.W))
  cycles := cycles +% 1.U

  printf("cycles %d\n", cycles)

  counter := cycles
}

object RegInitProblem {
  def build(optionsManager: TesterOptionsManager): String = {
    optionsManager.firrtlOptions =
      optionsManager.firrtlOptions.copy(compilerName = "low")

    chisel3.Driver.execute(optionsManager, () => new HasReg) match {
      case ChiselExecutionSuccess(Some(_), _, Some(firrtlExecutionResult)) =>
        firrtlExecutionResult match {
          case firrtl.FirrtlExecutionSuccess(_, compiledFirrtl) =>
            compiledFirrtl
          case firrtl.FirrtlExecutionFailure(message) =>
            throw new Exception(
              s"FirrtlBackend: Compile failed. Message: $message"
            )
        }
      case _ =>
        throw new Exception("Problem with compilation")
    }
  }

  def main(args: Array[String]): Unit = {
    // fixed the use of copy here.
    val optionsManager = new TesterOptionsManager {
//Line changed here
      treadleOptions = treadleOptions.copy(callResetAtStartUp = true, writeVCD=true)
    }
    val compiledFirrtl = build(optionsManager)
    val simulator = TreadleTester(compiledFirrtl, optionsManager)
    var done = false
    
    // example of how to look at treadle's view of the symbol table
    simulator.engine.symbolTable.nameToSymbol.keys.toSeq.sorted.foreach { key =>
      println(s"symbol: $key")
    }
    while (!done) {
      simulator.step(1)
      if (simulator.peek("counter") == 14) {
        done = true
      }
    }
    // this will print the status string
    simulator.report()
    // this will not print the status string, but will return if the tests are successful or not
    val isSuccess = simulator.finish()
  }
}
```

claford-v-lawrence, 2020-04-02:
I'm leaving this issue open because it's kind of unintuitive? To get an vcd file, you must both enable the `writeVCD=true` and call a `simulator.report` or `simulator.finish` in the end. Shouldn't setting the value alone or execute a function call alone be enough?

chick, 2020-04-02:
I think that's fine. I've been doing some background work on the VCD emitter and one of the things I would like to implement is to make it able to at least optionally do streaming output.

claford-v-lawrence, 2020-04-03:
OK, that explains the options and the function call. Thanks you for your clarification!

closed 2020-04-03 by claford-v-lawrence