chipsalliance/treadle #186: Using treadle to simulate rocket

davidmlw, 2020-04-01:
Using treadle to simulate rocket has benifits:
1. Fast edit-result feedback, as chisel/firrtl/treadle flow needs no firrtl->verilog emit and verilog->exe compilation, it could help user get result more quick.
2. Remove verilog EDA simulation tools dependency. Verilator provides a good alternative, it is good to have another one.
3. Fast simulation. As high firrtl remains more high-level information, treadle could use this information to get a chance to speedup. For example, if and when statements, treadle could avoid a bunch of sub-circuit execution when it observe false conditions, while a gate/verilog level simulation may have to exercise the false path.
4. Rocket is a large design now, the project can also develop new opportunities to accelerate the simulation, for example, circuit partition and multi-threading simulation for multi-rocket-core systems.

I am trying to get Treadle to simulate rocket.fir, and met one problem: rocket has to depend on some blackboxes, at least:
   - ClockGate https://github.com/chipsalliance/rocket-chip/blob/master/src/main/scala/util/ClockGate.scala, 
    The ClockGate is easy to workaround to provide another p(ClockGateImpl) with some modification of rocket-chip.

   - plusarg_reader https://github.com/chipsalliance/rocket-chip/blob/master/src/main/scala/util/PlusArg.scala
plusarg is a missing mechanism of Chisel and Treadle. Plusargs are well accepted by verilog EDA tools, and Chisel/Treadle could provide an elegant implementation. I just filed an Issue https://github.com/freechipsproject/chisel3/issues/1396 and https://github.com/freechipsproject/treadle/issues/185

With these improvements, using treadle to simulate rocket is feasible.

chick, 2020-04-01:
I had started building out he scala black boxes a while ago. l'll try and dig up what I had done.

davidmlw, 2020-04-02:
Nice. Will use

closed 2020-04-02 by davidmlw