chipsalliance/treadle #202: Is there an option to initialize all the register with random number?

davidmlw, 2020-05-15:
For the register without Init values, the firrtl spec is to initialize them with random number like real circuits. Current treadle seems to initialize all registers with zeros, it this true? Is there an option to initialize all the register with random number?

chick, 2020-05-15:
@davidmlw not currently but I think this would be very easy to implement in treadle. Is there any particular strategy, I could put a random (correctly sized) value in every symbols, or could constrain the randomizing to just registers or other specific kinds of symbols. Perhaps there would be some separate rules for memories

chick, 2020-07-15:
@davidmlw Please take a look at PR #215 which attempts to do this kind of randomization. It would be great if you could do a quick review

chick, 2020-12-30:
Closing because I think #215 addresses this

closed 2020-12-30 by chick