chipsalliance/treadle #204: How to peek and poke an internal signal or memory?

davidmlw, 2020-05-22:
I searched and didn't find an example.

```
peek("instA.instB.data", 42)
poke("instA.instB.data")
// or 
peek("instA_instB_data", 42)
poke("instA_instB_data")
```

davidmlw, 2020-05-22:
I saw an example here https://github.com/freechipsproject/treadle/blob/bb72c8c34a1ff707720956c21715d3c202c7bec4/src/test/scala/treadle/MemoryUsageSpec.scala#L420-L429, using `LoadMemoryAnnotation`, which uses `ComponentName`.

While I didn't find an example for `ComponentName` to reference a deep component.

sequencer, 2020-05-22:
try `ReferenceTarget`, I think `ComponentName` has been deprecated?

davidmlw, 2020-05-23:
```
peek("instA.instB.data", 42)
poke("instA.instB.data")
```
should work. I verified with a small example.

ComponentName will be transformed to ReferenceTarget.

chick, 2021-06-24:
I think this can be closed

closed 2021-06-24 by chick