chipsalliance/treadle #205: Some memory initialization option proposals

davidmlw, 2020-05-24:
https://github.com/freechipsproject/treadle/blob/1c67bc846aafc3bdd707f76ead8cefd5f93e0376/src/main/scala/treadle/executable/Memory.scala#L618-L663

1. One may provide the memory initial data directly from Scala data structures. It might be useful to provide LoadMemoryAnnotation from Seq[BigInt]. With this option, one can write any load memory annotations, for example elf loader, bin loader.

2. Some Memories may have a type of Vec(n, UInt()), that would make a Vec of memory instances. The row line needs be devided to n slices and load each of the memroy instances. If item 1 in place, one can split the file contents to `n` slices and load to each memory instances.

3. Treadle shellOptions could generate the LoadMemoryAnnotation. With the above features, the users can write their own shell options for their projects. For example:
`--init-memory circuit.module.component:8x4 --hexfile file`.

I can provide a user level example for item 3 if item 1,2 are in place.

ekiwi-sifive, 2020-06-15:
Treadle can probably reuse the `MemoryInitAnnotation` once it is merged in firrtl: https://github.com/freechipsproject/firrtl/pull/1645

ekiwi-sifive, 2020-07-14:
checkout https://github.com/freechipsproject/treadle/pull/220

closed 2021-04-29 by ekiwi