chipsalliance/treadle #219: Is it possible to access signals in clockChange?

davidmlw, 2020-07-09:
In current example, the users need maintain the internal states, the input/output update events. Is it possible to put all this in a single function `clockChange`? 

for example, `input` could be `peek`, `output` could be `poke`. If this kind of synchronized clocked ScalaBlackBox is possible. It would be very useful for many verification cases. Verification doesn't need to be at combinational boundary,  only at clock boundary is already very usefule :
```
override def clockChange(transition: Transition, clockName: String): Unit = {
     output("io_d") = input("io_d") + 1
     output("io_d2") = input("io_d1") + input("io_d2")
}
```

https://github.com/freechipsproject/treadle/blob/962acf6f7b5fbca3fe17ebe2897dcd48d6c465a1/src/test/scala/treadle/asyncreset/AsyncResetBlackBox.scala#L26-L56

chick, 2020-07-10:
I think the first part of the answer is yes. You can put as much logic in the clockChange method as you would like. It would be relatively simple I think to provide a hook to the treadle `ExecutionEngine` which would let black boxes access treadle's `DataStore` this could be dangerous of course. I think it would also be nice to allow black box creation to request some `DataStore` variables of it's own. That way internal state of black boxes would be retained in the snapshot mechanism. I think mostly what this needs is prioritization

davidmlw, 2020-07-12:
I see. It is dangerous to open whole access to treadle's `DataStore`. It is reasonable to open the ScalaBlackBox's input and output port for default, while other internal states need other scheme to get access. I take Github as a open discussion panel for records. Please make development decisions based on available resources. I am reading treadle code and will make further investigation some time.