chipsalliance/treadle #227: How to call `finish` from a `Module` or `ScalaBlackBox` ?

davidmlw, 2020-07-21:
It is a basic task to call `finish` from a `Module` in verilog, as it is "designed for simulation". While Chisel is designed for RTL elaboration, it seems Chisel lacks simulation control primitives. I think Treadle could contribute more in this area, by openning more internal capabilities to ordinary Modules.

I inspect a little for work around from `ScalaBlackBox`. But BlackBoxes depends on Annotations, and `TreadleTester` depends on Annotation to create, I have no idea how to pass a created TreadleTester instance handler to an annotation and then pass to itself.

https://github.com/freechipsproject/treadle/blob/e0d609fe4f65783ed4b8b66bcb244ae353570f50/src/test/scala/treadle/MemoryVcdSpec.scala#L84-L99

ekiwi, 2020-07-21:
If all you are trying to do is to stop the simulation when a condition becomes true on a rising clock edge, try out the `stop` statement. It can be used like this:

```.scala
when(stop_condition) { stop() }
```

and it translates to a Verilog `finish` call.

davidmlw, 2020-07-21:
Is this also supported by treadle and cause treadle finish current simulation?

chick, 2020-07-21:
Yes, In treadle when the stop connection is me treadle throws a StopException. This is handled properly by chisel-iotesters and chiseltest

davidmlw, 2020-07-22:
Oh, I see, Treadle uses Exception to notify the env, wise solution.

closed 2020-07-22 by davidmlw

davidmlw, 2020-07-22:
Some recommendations after using `stop`:
1. Chisel and Treadle could have another semantic `finish`. `finish` has different meanings and using scenario other than `stop`.
2. `TreadleTester.report` could take the user-intended finish as a success, or just take Linux status tradition, return code 0 is verything being OK.
3. On the contrary, the simulation is run out time, with return code `None`, should be seen as a failure.
4. Treadle can pass the circuit finish code to env, so other tools in flow can integrate it as a common Linux executables