chipsalliance/treadle #233: Seeking some advices of how to implement $fopen $write $close even in ScalaBlackBox.

davidmlw, 2020-08-04:
Example verilog code could be like below. One major concern would be what is the right time/place to call `file.close()` in a `ScalaBlackBox`. `initial` could be in class constructor, `@ posedge clock` could be in `clockEdge`, while I didn't find a good time for `final`.

```
module FileWriter #(parameter FILENAME="", DATAWIDTH=32) (
  input        clock,
  input        reset,
  input        valid,
  input [DATAWIDTH-1:0] data);

  integer f;

  initial begin
    f = $fopen(FILENAME, "w");
  end

  final begin
    $fclose(f);
  end

  always @(posedge clock) begin
    if (!reset && valid) begin
      $fwrite(f, "%x\n", data);
    end
  end

endmodule
```

chick, 2020-08-04:
Take a look and try out Implement a finish call for engine PR #234 as a possible implementation

chick, 2020-12-30:
Closing per comments in #234 
Re-open if necessary

closed 2020-12-30 by chick