chipsalliance/treadle #260: TODO: Unhandled statement Verification

danielkasza, 2020-10-18:
When running test cases using `chiseltest` on a module that uses `chisel-formal`, I am getting a lot of these messages:

```
TODO: Unhandled statement Verification(cover, @[Formal.scala 22:21],Reference(clock,ClockType,PortKind,SourceFlow),Reference(_T_2,UIntType(IntWidth(1)),NodeKind,SourceFlow),Reference(_T,UIntType(IntWidth(1)),NodeKind,SourceFlow),StringLit())
TODO: Unhandled statement Verification(assert, @[Formal.scala 23:11],Reference(clock,ClockType,PortKind,SourceFlow),Reference(_T_4,UIntType(IntWidth(1)),NodeKind,SourceFlow),DoPrim(and,List(Reference(_T_2,UIntType(IntWidth(1)),NodeKind,SourceFlow), Reference(_T,UIntType(IntWidth(1)),NodeKind,SourceFlow)),List(),UIntType(IntWidth(1))),StringLit())
TODO: Unhandled statement Verification(assert, @[StageHelper.scala 136:11],Reference(clock,ClockType,PortKind,SourceFlow),Reference(_T_8,UIntType(IntWidth(1)),NodeKind,SourceFlow),Reference(_T,UIntType(IntWidth(1)),NodeKind,SourceFlow),StringLit())
TODO: Unhandled statement Verification(assert, @[StageHelper.scala 143:11],Reference(clock,ClockType,PortKind,SourceFlow),Reference(_T_14,UIntType(IntWidth(1)),NodeKind,SourceFlow),Reference(_T,UIntType(IntWidth(1)),NodeKind,SourceFlow),StringLit())
```

I believe these come from Treadle: https://github.com/freechipsproject/treadle/search?q=TODO%3A+Unhandled+statement
It looks like Treadle does not support the Verification statements yet.

chick, 2020-10-19:
Thanks for reporting. I am looking to make a fix as soon as I can

chick, 2020-12-30:
PR #261 stops Treadle from giving these error messages.
It currently ignores these statements
I will be working on making it do something more useful with them in the near future

closed 2020-12-30 by chick