chipsalliance/treadle #291: UInt padding error

manageryzy, 2020-12-18:
```scala
class TM extends Module{
  val io = IO(new Bundle() {})

  val a = WireInit(1.U(2.W))
  val b = WireInit(2.U(3.W))

  val c = Wire(UInt(10.W))
  c := a -& b

  printf(p"a: ${Binary(a.asUInt())}\n")
  printf(p"b: ${Binary(b.asUInt())}\n")
  printf(p"c: ${Binary(c.asUInt())}\n")

  dontTouch(a)
  dontTouch(b)
  dontTouch(c)
}
```

outputs
```
a:  1
b:  10
c:         -1
```

in this case , unsigned padding was treated as siged. this is a bug of  expression compiler.  ExpressionCompiler do  substrate/assign directlly and ignored padding operate. **node c** was assgned raw internal data staorage value.

firrtl attached here.

```firrtl
circuit TM :
  module TM :
    input clock : Clock
    input reset : UInt<1>
  
    node a = UInt<2>("h1")
    node b = UInt<3>("h2")
    node _GEN_0 = pad(a, 3) @[Adder.scala 417:10]
    node _T = sub(_GEN_0, b) @[Adder.scala 417:10]
    skip
    skip
    node _T_3 = not(reset) @[Adder.scala 419:9]
    skip
    skip
    skip
    skip
    node c = pad(_T, 10) @[Adder.scala 416:15 Adder.scala 417:5]
    reg _GEN_1 : UInt<1>, clock with :
      reset => (UInt<1>("h0"), UInt<1>("h0")) @[Adder.scala 419:9]
    _GEN_1 <= _T_3 @[Adder.scala 419:9]
    reg _GEN_2 : UInt<2>, clock with :
      reset => (UInt<1>("h0"), UInt<1>("h0")) @[Adder.scala 419:9]
    _GEN_2 <= a @[Adder.scala 419:9]
    reg _GEN_3 : Clock, clock with :
      reset => (UInt<1>("h0"), UInt<1>("h0")) @[Adder.scala 419:9]
    _GEN_3 <= clock @[Adder.scala 419:9]
    reg _GEN_4 : UInt<1>, clock with :
      reset => (UInt<1>("h0"), UInt<1>("h0")) @[Adder.scala 419:9]
    _GEN_4 <= _T_3 @[Adder.scala 419:9]
    printf(_GEN_3, _GEN_1, "a: %b\n", _GEN_2) @[Adder.scala 419:9]
    reg _GEN_5 : UInt<1>, clock with :
      reset => (UInt<1>("h0"), UInt<1>("h0")) @[Adder.scala 420:9]
    _GEN_5 <= _T_3 @[Adder.scala 420:9]
    reg _GEN_6 : UInt<3>, clock with :
      reset => (UInt<1>("h0"), UInt<1>("h0")) @[Adder.scala 420:9]
    _GEN_6 <= b @[Adder.scala 420:9]
    reg _GEN_7 : Clock, clock with :
      reset => (UInt<1>("h0"), UInt<1>("h0")) @[Adder.scala 420:9]
    _GEN_7 <= clock @[Adder.scala 420:9]
    reg _GEN_8 : UInt<1>, clock with :
      reset => (UInt<1>("h0"), UInt<1>("h0")) @[Adder.scala 420:9]
    _GEN_8 <= _T_3 @[Adder.scala 420:9]
    printf(_GEN_7, _GEN_5, "b: %b\n", _GEN_6) @[Adder.scala 420:9]
    reg _GEN_9 : UInt<1>, clock with :
      reset => (UInt<1>("h0"), UInt<1>("h0")) @[Adder.scala 421:9]
    _GEN_9 <= _T_3 @[Adder.scala 421:9]
    reg _GEN_10 : UInt<10>, clock with :
      reset => (UInt<1>("h0"), UInt<1>("h0")) @[Adder.scala 421:9]
    _GEN_10 <= c @[Adder.scala 421:9]
    reg _GEN_11 : Clock, clock with :
      reset => (UInt<1>("h0"), UInt<1>("h0")) @[Adder.scala 421:9]
    _GEN_11 <= clock @[Adder.scala 421:9]
    reg _GEN_12 : UInt<1>, clock with :
      reset => (UInt<1>("h0"), UInt<1>("h0")) @[Adder.scala 421:9]
    _GEN_12 <= _T_3 @[Adder.scala 421:9]
    printf(_GEN_11, _GEN_9, "c: %b\n", _GEN_10) @[Adder.scala 421:9]
```

chick, 2020-12-30:
Thanks for the report. I'm working on a fix

chick, 2020-12-30:
I think I've fixed this in PR #292 
Please take a look if you have time

closed 2021-02-18 by ekiwi