chipsalliance/treadle #306: Add support for SAIF dump

hansemandse, 2021-03-05:
Hi guys,

Just an idea; have you considered adding support for dumping SAIF results? For some rather lengthy testers2 tests, VCD dumps become huge, so being able to retrieve SAIF results for power estimation instead would be great.

chick, 2021-03-05:
This has come up before, seems like an ok idea.
[SAIF spec, just for reference](https://link.springer.com/content/pdf/bbm%3A978-1-4614-4271-4%2F1.pdf)
Chiseltest relies on the backend to write the VCD, I don't think verilator will write SAIF, yet it is typically the backend used for larger or longer running tests.
Treadle could be pretty easily modified to add SAIF output but since it is slower and typically used on smaller designs I don't know whether it would be worth it. 
There is a branch of Treadle (streaming-vcd) that has a streaming VCD option which can speed up the VCD writing, the current behavior is to save up all changes in memory and then write it all out at the end. Streaming should be faster for long runs.
The version that branch is based on is a bit old but it should be pretty easy to port forward.

hansemandse, 2021-03-06:
I see. As you mention, I am indeed using Verilator for my larger tests, but I typically get memory warnings near the end of long tests if I don't manually allow SBT to use more than its default amount. 
I can't seem to find any mention of SAIF in the Verilator doc, and I don't suppose it makes sense to include it in Treadle only.
For now, I will have a look at the streaming-vcd version instead. 

Thanks for helping! Feel free to close this, unless you want to keep it open :)

ekiwi, 2021-03-10:
We have a feature request (from me) for FST support for treadle which would allow us to get FST from both treadle and Verilator: https://github.com/chipsalliance/treadle/issues/101