chipsalliance/treadle #321: dshr sometimes incorrectly computed

TheAIBot, 2021-04-28:
Treadle uses int and long shifts if the wire width is small enough. Problem with doing this is that some computer architectures do not behave as one would expect. Given ```a >> b``` where both a and b are 32bit ints, x86 will only use the 5 lsb of ```b``` to figure out how much ```a``` should be shifted by. if ```a``` is a 64bit int then the 6 lsb are used instead.

This effectively means that shifting beyond 31 is incorrect when a wires width is small enough and therefore uses [DshrInts](https://github.com/chipsalliance/treadle/blob/master/src/main/scala/treadle/executable/IntOps.scala#L140) to shift with. Same is true for [DshrLongs](https://github.com/chipsalliance/treadle/blob/master/src/main/scala/treadle/executable/LongPrimOps.scala#L115) when shifting more than 63.

Table below shows what values treadle gives when right shifting 1 by different amounts. Included [DshrBigs](https://github.com/chipsalliance/treadle/blob/master/src/main/scala/treadle/executable/BigPrimOps.scala#L115) to show what the correct value should be.

| dshr(1, shift) | DshrInts | DshrLongs | DshrBigs |
|---------------|----------|--------------|-----------|
| shift = 31      |            0 |                0 |             0 |
| shift = 32      |            1 |                0 |             0 |
| shift = 33      |            0 |                0 |             0 |
| shift = 63      |            0 |                0 |             0 |
| shift = 64      |            1 |                1 |             0 |
| shift = 65      |            0 |                0 |             0 |

Below is the code that generated the data used in the table.

```scala
class ShiftTestInt extends Module {
    val io = IO(new Bundle() {
        val a = Input(UInt(7.W))
        val b = Input(UInt(7.W))
        val dshrOut = Output(UInt(7.W))
    })
    io.dshrOut := io.a >> io.b
}

class ShiftTestLong extends Module {
    val io = IO(new Bundle() {
        val a = Input(UInt(37.W))
        val b = Input(UInt(19.W))
        val dshrOut = Output(UInt(37.W))
    })
    io.dshrOut := io.a >> io.b
}

class ShiftTestBigInt extends Module {
    val io = IO(new Bundle() {
        val a = Input(UInt(65.W))
        val b = Input(UInt(19.W))
        val dshrOut = Output(UInt(65.W))
    })
    io.dshrOut := io.a >> io.b
}

class ShiftTester extends FlatSpec with ChiselScalatestTester with Matchers {
    it should "Test shift with FuncInt" in {
        test(new ShiftTestInt).withFlags(Array("--tr-write-vcd")) {
            dut=> {
                dut.io.a.poke(1.U)

                dut.io.b.poke(31.U)
                dut.clock.step()

                dut.io.b.poke(32.U)
                dut.clock.step()

                dut.io.b.poke(33.U)
                dut.clock.step()

                dut.io.b.poke(63.U)
                dut.clock.step()

                dut.io.b.poke(64.U)
                dut.clock.step()

                dut.io.b.poke(65.U)
                dut.clock.step()
            }
        }
    }
    it should "Test shift with FuncLong" in {
        test(new ShiftTestLong).withFlags(Array("--tr-write-vcd")) {
            dut=> {
                dut.io.a.poke(1.U)

                dut.io.b.poke(31.U)
                dut.clock.step()

                dut.io.b.poke(32.U)
                dut.clock.step()

                dut.io.b.poke(33.U)
                dut.clock.step()

                dut.io.b.poke(63.U)
                dut.clock.step()

                dut.io.b.poke(64.U)
                dut.clock.step()

                dut.io.b.poke(65.U)
                dut.clock.step()
            }
        }
    }
    it should "Test shift with FuncBig" in {
        test(new ShiftTestBigInt).withFlags(Array("--tr-write-vcd")) {
            dut=> {
                dut.io.a.poke(1.U)

                dut.io.b.poke(31.U)
                dut.clock.step()

                dut.io.b.poke(32.U)
                dut.clock.step()

                dut.io.b.poke(33.U)
                dut.clock.step()

                dut.io.b.poke(63.U)
                dut.clock.step()

                dut.io.b.poke(64.U)
                dut.clock.step()

                dut.io.b.poke(65.U)
                dut.clock.step()
            }
        }
    }
}

```

chick, 2021-04-29:
@TheAIBot Great catch. Thanks for the great example. I have created PR #322 to address this.
If you have time, please take a look at the PR too see if you think it is sufficient

TheAIBot, 2021-04-29:
@chick pr looks good, should solve the problem. Thanks you for resolving the issue so quickly 👍

closed 2021-04-29 by TheAIBot