chipsalliance/treadle #352: Make it possible to create a TreadleTester without running firrtl optimizations

ekiwi, 2021-08-23:
It would be nice if we could allow users to test an un-optimized circuit.

This shouldn't be too hard to do, the major question is how to best expose this option to the user.

chick, 2022-02-23:
Is this just mean changing `Forms.LowFormOptimized` to `Forms.LowForm` in `PrepareAst`

ekiwi, 2022-02-23:
Potentially. We would have to play around with this a little bit.

chick, 2022-02-25:
MIght as well add this in too. Treadle's transformations should be idempotent. Which it is not right now, the Augment printf will keep adding registers in between the clock, enable, and arguments each time it is run.
Actually. at the moment, is not even possible because the registers generated by Augment do not have correctly sized `init` values, which causes failures in `CheckTypes`, this should be fixed.