chipsalliance/treadle #407: Incompatibility with `json4js-native` 4.0.5

michael-etzkorn, 2022-04-22:
FIRRTL updated to 4.0.5 today so the 1.6-SNAPSHOT of Treadle is broken. 
At this commit: https://github.com/chipsalliance/firrtl/commit/f31416a8b7122ab29b3243c5d35670030e244455

Chiseltest failure on main: 

[image]

I suspect somewhere around here is the incompatibility with the new major version:
https://github.com/chipsalliance/treadle/blob/master/src/main/scala/treadle/executable/DataStore.scala#L490-L501

sequencer, 2022-04-22:
Thanks for pointing out!
Fixing.

michael-etzkorn, 2022-04-22:
Thanks for the help Jiuyang! I was looking to see if I could fix it myself, but there's a fair amount of setup and brushing up I'll need to do to get my environment set up to regression test the fix. Chiseltest's `main` should be back to working as usual once we have a fix in.

sequencer, 2022-04-23:
strange. treadle test successfully with latest FIRRTL.

sequencer, 2022-04-23:
I doubt in your classpath, there exists another jackson/json4s, like in chipsalliance/rocket-chip#2968.

michael-etzkorn, 2022-04-24:
I believe even if it's compiling and running successfully with the newest json4s in internal tests, the jar `treadle_2.12-1.6-SNAPSHOT.jar` is stale and compiled with json4s 3.6 so it needs to be updated to work with the newest FIRRTL jar.

sequencer, 2022-04-24:
where did you introduced the json4s 3.6? AFAIK that was rocket-chip.

michael-etzkorn, 2022-04-25:
Rocket-chip shouldn't be relevant here. 

The facts as far as I understand them:
- json4s is introduced in FIRRTL. So when FIRRTL was still on 3.6 that's the last jar compiled for treadle.  
- Because json4s is a dependency in FIRRTL (which treadle depends on), json4s is not explicitly added as a dependency in Treadle's build. 
- Treadle makes use of the json4s package here: https://github.com/chipsalliance/treadle/blob/master/src/main/scala/treadle/executable/DataStore.scala#L6-L8 
- When FIRRTL upgraded to json4s 4.x, the treadle jar's compilation with 3.6 became stale. 
- It is likely possible to rectify this simply by recompiling and publishing the Treadle jar with a clean project since FIRRTL is now updated to json4s 4.x, this should fetch json4s 4.x and compile against that in treadle.

closed 2022-04-25 by jackkoenig