chipsalliance/treadle #415: syntax error(s) when parsing low Firrtl

wky17, 2022-08-24:
when I try to parser **Accumulator** low Firrtl code through treadle, 2 syntax error(s) detected.

the low Firrtl code:
```
val accuFirrtl1: String =
      s"""
         |circuit fib :
         |  module fib :
         |    input go : UInt<1>
         |    input reset : UInt<1>
         |    input clk : Clock
         |    output out : UInt<4>
         |
         |    reg n : UInt<4>, clk with :
         |      reset => (UInt<1>("h0"), n)
         |    node _T_11 = add ( n,go )
         |    node _T_12 = tail ( _T_11 , 1 )
         |    out <= n
         |    n <= mux(reset, UInt<4>("h0"), _T_12)
         |    """.stripMargin
```

when I try to parse it using: `TreadleTestHarness(Seq(FirrtlSourceAnnotation(accuFirrtl1)))`. the error is:
[image]

but when I rewrite **Accumulator** as:
```
val accuFirrtl: String =
      s"""
         |circuit fib :
         |  module fib :
         |    input go : UInt<1>
         |    input reset1 : UInt<1>
         |    input clk : Clock
         |    output out : UInt<4>
         |
         |    reg n : UInt<4>, clk with :
         |      reset => (UInt<1>("h0"), n)
         |    out <= n
         |    n <= mux(reset1, UInt<4>("h0"), tail(add(n, go),1))
         |    """.stripMargin
```

it's compiled successfully:
[image]

the only difference is that I unfold the `tail(add(n, go)` expression in the failed case, while I think it's proper expression in low Firrtl.
I wonder if treadle can parse this unfold expression.

chick, 2022-08-24:
I believe this is caused by the firrtl parser being very sensitive to spaces in some contexts.
by fixing up the use of spaces on the lines defining `_T_11` and `_T_12` the following string worked for me:
```
s"""
       |circuit fib :
       |  module fib :
       |    input go : UInt<1>
       |    input reset : UInt<1>
       |    input clk : Clock
       |    output out : UInt<4>
       |
       |    reg n : UInt<4>, clk with :
       |      reset => (UInt<1>("h0"), n)
       |    node _T_11 = add(n, go)
       |    node _T_12 = tail(_T_11 , 1)
       |    out <= n
       |    n <= mux(reset, UInt<4>("h0"), _T_12)
       |    """.stripMargin
```

wky17, 2022-08-29:
yes this worked, thank u so much!

closed 2022-08-29 by wky17